bitsy-ai/rpi-object-tracking #15: Proposing to include face detection together

Martin2kid, 2020-02-06:
* Raspberry Pi Deep PanTilt version: 1.0.1
* Python version: 3.7
* Operating System: debian
Pi4, 4GB, 64GB SD, Clean rpi-deep-pantilt install only on SD with .VENV setup

Improvement suggestion

When person is close to Picam, it tracks body & miss out face (face being out of screen frame) unless it started with face first & face is stay close to Picam without whole-body detected. 

It would be great to include face detection model into this project, so it's not only able to track person with face within the frame but also it can be used to identify person with further implementation. (like identify person from model first -- identify face from face model second --track face first or adjust Picam to fit face into center of frame & maintain face in the  center of frame)

Also note, new release & only available Coral USB Runtime 2.1 current version is not running with this rpi-deep-pantilt version & I couldn't find older version of Runtime to duplicate running this version with Coral USB.

Great project!

leigh-johnson, 2020-02-09:
Hey @Martin2kid! It turned out to be pretty easy to add `facessd_mobilenet_v2_quantized_320x320_open_image_v4`. 

I've included a FLOAT32 and UINT8 model in this PR, and uploaded the model [tarballs to the v1.0.1 release ](https://github.com/leigh-johnson/rpi-deep-pantilt/releases/tag/v1.0.1) if you'd like to inspect them. 

The Edge TPU performance is excellent on my RPI 4 (greater than 60 FPS, wow). 

I'll cut a release and upload to PyPi this afternoon.

leigh-johnson, 2020-02-09:
`pip install --upgrade rpi-deep-pantilt` 
`rpi-deep-pantilt face-detect` - detect all faces in the frame
`rpi-deep-pantilt face-track` - track faces

Both of these commands also support the `--edge-tpu` option as well. I know we're still trying to sort out installation over in https://github.com/leigh-johnson/rpi-deep-pantilt/issues/13. I'm going to spend some time trying to reproduce that issue.

closed 2020-02-09 by leigh-johnson

Martin2kid, 2020-02-10:
Wow! 
I certainly will & thank you very much!!!

Martin2kid, 2020-02-11:
Uninstalled older version by using pip uninstall rpi-deep-pantilt
New install using pip install rpi-deep-pantilt==1.1.0
Great!!! It detect face but rpi-deep-pantilt face-detect is much much slower than rpi-deep-pantilt detect. probably too much processing for Pi 4 by itself.

Thank you much!

leigh-johnson, 2020-02-11:
The FLOAT32 model is pretty slow, yeah. I'm seeing 30+ fps on the UINT8 quantized model with Edge TPU acceleration though. 

I'll see what I can do about fixing my usage to work with 13.0 and/or packaging a compatible version.

Martin2kid, 2020-02-12:
Thank you much!
I've experimented  OpenCV & Caffe for face detection in the past and it was resonably fast with Pi4, obviously it's totally different setup and I like Tensor's scalability.

I'll look forward.

Martin2kid, 2020-02-12:
Hmm,
It was running twice (rpi-deep-pantilt face-detect), now it throws error and Pan & Tilt is not working (rpi-deep-pantilt track is still working though).

Here is CMD output
(.venv) pi@raspberrypi:~/rpi-deep-pantilt $ rpi-deep-pantilt face-detect
INFO: Initialized TensorFlow Lite runtime.
^CException in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/rpi-deep-pantilt/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/camera.py", line 81, in render_overlay
    self.overlay.update(self.overlay_buff)
  File "/home/pi/rpi-deep-pantilt/.venv/lib/python3.7/site-packages/picamera/renderers.py", line 448, in update
    buf.data = source
  File "/home/pi/rpi-deep-pantilt/.venv/lib/python3.7/site-packages/picamera/mmalobj.py", line 1619, in _set_data
    with self as buf:
  File "/home/pi/rpi-deep-pantilt/.venv/lib/python3.7/site-packages/picamera/mmalobj.py", line 1725, in __enter__
    ct.POINTER(ct.c_uint8 * self._buf[0].alloc_size)).contents
OverflowError: cannot fit 'int' into an index-sized integer

Martin2kid, 2020-02-12:
I thought it may have something to do with buffer & ran;
sudo rasps-config
advanced--resolution--set to DMT Mode  31 1920*1080
 and 
sudo nano /boot/config.txt
set frame buffer width 1920 & frame buffer height to 1024

This is CMD output;
(.venv) pi@raspberrypi:~/rpi-deep-pantilt $ rpi-deep-pantilt face-detect
INFO: Initialized TensorFlow Lite runtime.
^CException in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/rpi-deep-pantilt/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/camera.py", line 81, in render_overlay
    self.overlay.update(self.overlay_buff)
  File "/home/pi/rpi-deep-pantilt/.venv/lib/python3.7/site-packages/picamera/renderers.py", line 447, in update
    buf = self.renderer.inputs[0].get_buffer()
  File "/home/pi/rpi-deep-pantilt/.venv/lib/python3.7/site-packages/picamera/mmalobj.py", line 1141, in get_buffer
    'cannot get buffer from disabled port %s' % self.name)
picamera.exc.PiCameraPortDisabled: cannot get buffer from disabled port vc.ril.video_render:in:0: Argument is invalid