bitsy-ai/rpi-object-tracking #20: Add --no-pantilt option to run model without the need for Pantilt module.

Grippy98, 2020-02-14:
* rpi-deep-pantilt version:
* Python version:
* TensorFlow version:
* Operating System:

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

leigh-johnson, 2020-02-14:
👋 Hey, thanks for the request. I think this is reasonable. 

Just wondering, what error do you get run you run the `detect` or `face-detect` commands without the Pan-tilt module?

cheewee2000, 2020-05-12:
I'd love to see this implemented as well! I'm building something that doesn't require camera movement.  I don't get any errors when I run detect or face-detect. I just get ...Initialized TensorFlow Light runtime and it just hangs there. No camera preview either.

When I run it will --loglevel=DEBUG I get the info printouts for framerate, but no camera preview.

cheewee2000, 2020-05-12:
I get these errors when I run 
$rpi-deep-pantilt test camera

Traceback (most recent call last):
  File "/home/pi/.venv/bin/rpi-deep-pantilt", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/cli.py", line 153, in main
    cli()
  File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/cli.py", line 149, in camera
    return camera_test()
  File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/control/hardware_test.py", line 13, in camera_test
    pantilthat.pan(0)
  File "/home/pi/.venv/lib/python3.7/site-packages/pantilthat/pantilt.py", line 466, in servo_one
    self.setup()
  File "/home/pi/.venv/lib/python3.7/site-packages/pantilthat/pantilt.py", line 72, in setup
    self._i2c = SMBus(1)
FileNotFoundError: [Errno 2] No such file or directory

PR #32 (2020-05-25): Make pantilt installation optional, merge face-detect and face-track into detect/track commands
closes #20
commit 1c47ba5c (2020-05-25): Make pantilt installation optional, merge face-detect and face-track into detect/track commands (#32)

* first pass on combining track/face-track and detect/face-detect into single CLI target with nargs labels

add refactor run_detect fns into camera.run_detect_stationary and
camera.run_detect_pantilt

* zero scores on labels not in provided label list for rpi-deep-pantilt detect [LABELS] command

* remove pan-tilt HAT zero from camera test (allows usage w/o pantilt installed) closes #20

* clean up CLI commands

* update README with merged detect/track command usage
commit 1dc9690d (2020-05-26): release v1.2.0 (#28)

* add a note to the README concerning /boot/config.txt (#22)

* Update README.md

* add a note about dtparam=i2c1=on in /boot/config.txt for Rasberry 4 Rasberian users

* update README.md with installation instructions for TensorFlow 2.2 (#30)

* --edge-tpu initializes Coral Interpreter packaged in tflite_runtime instead of TensorFlow.lite.interpreter.Interpreter (#31)

* Make pantilt installation optional, merge face-detect and face-track into detect/track commands (#32)

* first pass on combining track/face-track and detect/face-detect into single CLI target with nargs  [... cut]
closed 2020-05-26 by leigh-johnson