bitsy-ai/rpi-object-tracking #35: Error encountered when upgrading rpi-deep-pantilt

TCIII, 2020-06-05:
*pi-deep-pantilt version: 1.1.0
Python version: 3.7.3
TensorFlow version: 2.0.2
Operating System: Raspbian Buster Desktop on Rpi 4B+ w/4GB RAM

### Description
When I attempted to upgrade rpi-deep-pantilt to version 1.2.0, the upgrade failed.

### What I Did
pi@raspberrypi:~ $ python3 -m pip install --upgrade rpi-deep-pantilt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi-deep-pantilt
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/ed/9b/718d6851bef44dc333f56631333957948aa8b7b2fbbf98525b6ae070c4f7/rpi_deep_pantilt-1.2.0-py2.py3-none-any.whl
Collecting h5py (from rpi-deep-pantilt)
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Downloading https://www.piwheels.org/simple/h5py/h5py-2.10.0-cp37-cp37m-linux_armv7l.whl (4.7MB)
    100% |████████████████████████████████| 4.7MB 97kB/s 
Collecting tensorflow>=2.2.0 (from rpi-deep-pantilt)
  Could not find a version that satisfies the requirement tensorflow>=2.2.0 (from rpi-deep-pantilt) (from versions: 0.11.0, 1.12.0, 1.13.1, 1.14.0)
No matching distribution found for tensorflow>=2.2.0 (from rpi-deep-pantilt)

I assume that TensorFlow needs to be updated to version 2.2.0 or greater since that is what the rpi-deep-pantilt upgrade is requesting?

Regards,
TCIII

TCIII, 2020-06-05:
Okay, went back to the README.md and found this CLI string: pip install https://github.com/leigh-johnson/Tensorflow-bin/releases/download/v2.2.0/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl
which I successfully used to uninstall tensorflow 2.0.0 and associated components and install 2.2.0 and associated components.

I then was able to successfully upgrade rpi-deep-pantilt to 1.2.0 using: pip install --upgrade rpi-deep-pantilt without encountering any errors.

Regards,
TCIII

leigh-johnson, 2020-06-05:
Sorry that you had to dig for info @TCIII 

I'm going to re-open this issue as a reminder to add an explicit section on upgrading to the Readme.

TCIII, 2020-06-05:
@leigh-johnson,

No problem. I am good at Arduino C++, but still learning to deal with Python commands and programming so it was a good exercise for me.

Regards,
TCIII