bitsy-ai/rpi-object-tracking #36: Using the new Pi 12 mp HQ Camera in place of the Pi V2.1 8 Mp Camera

TCIII, 2020-06-14:
*pi-deep-pantilt version: 1.2.0
Python version: 3.7.3
TensorFlow version: 2.2.0
Operating System: Raspbian Buster Desktop on Rpi 4B+ w/4GB RAM

### Description
I am using the new Pi 12 Mp HQ Camera with the Pi 6 mm Lens, in place of the Pi V2.1 8 Mp Camera, on a custom servo gimbal that is functionally the same as the Pimoroni Pan/Tilt Gimbal.

### What I Did
First of all I used the rpi-deep-pantilt test camera program to verify that the Rpi 4B was seeing the new HQ Camera which is was and the video looked nice and sharp.

Then I proceeded to use the rpi-deep-pantilt test pantilt to very the performance of the pan/tilt gimbal. I had to reverse the tilt value to get the tilt gimbal to move in correct direction due to my custom gimbal setup.

My next action was to use the rpi-deep-pantilt detect to verify that the program could detect and classify me which it did. Since the 6 mm Lens is wide angle, I could move over quite a distance horizontally before I exited the detection box and the FOV.

Finally I ran the rpi-deep-pantilt track to see if the program could successfully track me if I began to move out of the FOV. I found that I had to reverse both the pan and tilt directions to get the program to successfully track me. I am using the default PID values and will have to adjust them for my custom gimbal setup. The pan gimbal moved smoothly, but slowly. The tilt gimbal moved slowly and did hunt slightly after the detect box had captured me and I probably will have to adjust the I and D PID values.

Bottom line here is that the new Pi 12 Mp HQ Camera works well with Leigh's programs, however should the frame size be adjusted for the larger 12 Mp chip and the wider FOV? Suggestions welcomed!

```

leigh-johnson, 2020-06-25:
Thanks for the detailed report @TCIII! I just got my hands on one of the new cameras, so I'll see if I can reproduce this issue and push a fix. 

Just curious, can you tell me more about your gimbal setup? I've been looking into alternatives for Pimoroni's HAT and would love to support a variety of configurations. =) 

The "seeking" routine in my PID controller leaves a lot to be desired. I'm considering writing a calibration routine for it!  Right now the default values assume the camera is mounted at desk height, which doesn't support real-world applications where the camera is mounted higher/lower.

leigh-johnson, 2020-06-25:
If it helps in the meantime, here are my notes on PIDs in general. You're spot-on that the integrated historical values and derivative "dampening" are the values to tweak for motion "smoothing". I suspect you'll also need to set different P values on the X and Y axis for a wide-angle lens - the X value should be smaller than the Y value to compensate for a wider field of view on the X axis. 

https://miro.medium.com/max/1400/1*9sQpP7SqMHrhAwySJBJffQ.png

TCIII, 2020-06-25:
Hi Leigh,
Thanks for the timely response, much appreciated.
I started out with the Pimoroni HAT to get your project working, but decided to step up to bigger gimbals to support bigger cameras such as the ELP and Spinel webcams that can support various lens assemblies.
Here is what I did:
1) Since the Rpi really cannot provide enough drive current for bigger servos I bought two more Pimornoi HATs without the servo gimbals. The objective here is to provide an outboard +5 vdc source to drive the bigger servos.
2) I then attached two female to male servo cable extenders to the Servo 1 and Servo 2 Pimoroni HAT connectors. I presently have my Rpi 4B mounted in a Miuzei case (https://www.amazon.com/gp/product/B07VX3HQGJ/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) which includes a fan. I used a Stacking Header (https://www.amazon.com/gp/product/B01IRRCEBK/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1) to raise the Pimoroni HAT above the top of the Miuzei case to allow the servo cables to clear the case fan at the top of the case.
3) Finally I then fabbed up a small general purpose pwb that accepts the pan and tilt servo pwm and grounds signals, but not the +5 vdc from the Rpi 4B, and provides the necessary servo signals to drive the bigger pan and tilt gimbals. The +5 vdc power is provided by an outboard power supply capable of driving the bigger servos.
4) My bigger pan and tilt gimbal assembly is comprised of a Hitec HS-422 standard servo for the pan gimbal and a Hitec HS-645 MG standard servo for the tilt servo. The two servos are mounted in this Lynxmotion gimbal assembly 9https://www.robotshop.com/en/lynxmotion-pan-and-tilt-kit-aluminium2.html). The pan servo can be mounted to a base using this servo bracket (https://www.robotshop.com/en/lynxmotion-aluminum-servo-bracket.html) or you can use this Base Rotate Kit   (https://www.robotshop.com/en/lynxmotion-base-rotate-kit-hs-422-servo.html) and the servo bracket to mount the tilt servo gimbal to the base rotate kit. Lynxmotion assembly instructions here: http://www.lynxmotion.com/images/html/build153.htm

In actuality I prefer using the Adafruit 16-Channel 12-bit PWM/Servo Driver - I2C interface - PCA9685 in place of the Pimoroni HAT which uses a proprietary IC chip. The PCA9685 is used the world over to drive servos using a microprocessor. An example of its use can be found in this Paul McWhorter servo gimbal tutorial here https://www.youtube.com/watch?v=5WeTAA8Unqo&list=PLGs0VKk2DiYxP-ElZ7-QXIERFFPkOuP4_&index=30 and here https://www.youtube.com/watch?v=8YKAtpPSEOk&list=PLGs0VKk2DiYxP-ElZ7-QXIERFFPkOuP4_&index=31. As you can see, Adrian isn't the only one doing pan and tilt detection and tracking.

A servo gimbal calibration routine sounds great as Adrian essentially had the user calibrate each gimbal axis one at a time using the main track program which was kind of hit and miss and very tedious. Your present values work great for the Pimoroni micro servos, but are obviously not suited for bigger, faster servos with different inertias. I will be glad to work with you to proof out and debug any servo gimbal calibration code as I am retired and not going anywhere at the moment.

I will take a look at your notes on PID tuning as I am always interested in learning new processes.

I would like to use your project to detect and track aircraft and UFOs (don't laugh) in the area I live in in Central, FL. There have been several Discovery Channel Josh Gates episodes concerning the detection and tracking of UFOs: https://go.discovery.com/tv-shows/expedition-x/full-episodes/mt-adams-ufo-encounter If you can watch the video you will notice that the MUFON UFO hunter had a Rpi 3B+ with an environmental sensing HAT in his UFO tracker along with a very expensive ($1,000) PTZ Dahua surveillance camera. 

Regards,
TCIII