bitsy-ai/rpi-object-tracking #37: Error encountered when trying to use Google Coral accelerator

GBW48, 2020-06-24:
* rpi-deep-pantilt version: v1.2.0
* Python version: 3.7.3
* TensorFlow version: 2.2
* Operating System: Raspbian 10 (buster), running on a 4 gig Raspberry Pi 4

### Description

Object detection using Google Coral accelerator (following README.md instructions for rpi-deep-pantilt version 1.2) fails and does not display image.  Please note standard object detection without using the accelerator works ("rpi-deep-pantilt detect --rotation 180").  I repeated the deep-pan-tilt build and object detection twice -- with the same result.

When I issue "rpi-deep-pantilt detect --edge-tpu --rotation 180", I get two errors with no image displayed on the RPi.

Fist error:
> Traceback (most recent call last):
>   File "/home/pi/.venv/bin/rpi-deep-pantilt", line 10, in <module>
>     sys.exit(main())
>   File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/cli.py", line 172, in main
>     cli()
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 829, in __call__
>     return self.main(*args, **kwargs)
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 782, in main
>     rv = self.invoke(ctx)
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 610, in invoke
>     return callback(*args, **kwargs)
>   File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/cli.py", line 96, in detect
>     run_stationary_detect(labels, model_cls, rotation)
>   File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/camera.py", line 80, in run_stationary_detect
>     model = model_cls()
>   File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/ssd_mobilenet_v3_coco.py", line 65, in __init__
>     tf.lite.experimental.load_delegate(self.EDGETPU_SHARED_LIB)
>   File "/home/pi/.venv/lib/python3.7/site-packages/tensorflow/lite/python/interpreter.py", line 161, in load_delegate
>     delegate = Delegate(library, options)
>   File "/home/pi/.venv/lib/python3.7/site-packages/tensorflow/lite/python/interpreter.py", line 90, in __init__
>     self._library = ctypes.pydll.LoadLibrary(library)
>   File "/usr/lib/python3.7/ctypes/__init__.py", line 434, in LoadLibrary
>     return self._dlltype(name)
>   File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
>     self._handle = _dlopen(self._name, mode)
> OSError: libedgetpu.so.1: cannot open shared object file: No such file or directory
> Exception ignored in: <function Delegate.__del__ at 0xa5f3bfa8>
> 

Second error:
> Traceback (most recent call last):
>   File "/home/pi/.venv/lib/python3.7/site-packages/tensorflow/lite/python/interpreter.py", line 125, in __del__
>     if self._library is not None:
> AttributeError: 'Delegate' object has no attribute '_library'

### What I Did
I freshly installed Raspbian 10 (buster) on a high speed 32 gig SD card.  I then followed the README.md instructions.  I only had one issue after issuing the "pip install https://github.com/leigh-johnson/Tensorflow-bin/releases/download/v2.2.0/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl" command.  

Here is the error:
> Building wheels for collected packages: grpcio
>   Running setup.py bdist_wheel for grpcio ... error
>   Complete output from command /home/pi/.venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-tjyl9jcz/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-okaw4p5s --python-tag cp37:
>   Found cython-generated files...
>   usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
>      or: -c --help [cmd1 cmd2 ...]
>      or: -c --help-commands
>      or: -c cmd --help
> 
>   error: invalid command 'bdist_wheel'
> 
>   ----------------------------------------
>   Failed building wheel for grpcio
>   Running setup.py clean for grpcio
> Failed to build grpcio
> 

Otherwise, I encountered no problems with the install.  Please note, I was careful to make all recommended changes config.txt and Raspberry Pi interfaces.

leigh-johnson, 2020-06-25:
Thanks for the bug report @GBW48, sorry this didn't work for you!

Looks like your TensorFlow installation didn't finish because you don't have the `wheel` Python package installed. I think my Coral installation instructions are out of date too. Try the following to re-install, let me know how it goes. 🤞 

Let me know how it goes a (with your virtual environment active) to reinstall everything

1. Install `wheel` and reinstall TensorFlow

`(.venv)$ pip install wheel && pip install --force-reinstall https://github.com/leigh-johnson/Tensorflow-bin/releases/download/v2.2.0/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl`

2. Install/reinstall Coral's libedgetpu.  

```bash
$ echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list

$ curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install libedgetpu1-std
```

3. Re-install Coral's tflite_runtime

`(.venv)$ pip install https://dl.google.com/coral/python/tflite_runtime-2.1.0.post1-cp37-cp37m-linux_armv7l.whl`

GBW48, 2020-06-25:
Hi Leigh, Thanks so much for the quick response.  With your suggested changes, I am still getting run-time errors and I'm unable to display objects using the Coral accelerator.  

**I tried your suggestions on a fresh build.  The wheel and TensorFlow installation appeared to work.  Here are the results of the install.**
> (.venv) pi@raspberrypi:~ $ pip install wheel
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting wheel
>   Downloading https://files.pythonhosted.org/packages/8c/23/848298cccf8e40f5bbb5  9009b32848a4c38f4e7f3364297ab3c3e2e2cd14/wheel-0.34.2-py2.py3-none-any.whl
> Installing collected packages: wheel
> Successfully installed wheel-0.34.2
> (.venv) pi@raspberrypi:~ $
> (.venv) pi@raspberrypi:~ $ pip install --force-reinstall https://github.com/leig  h-johnson/Tensorflow-bin/releases/download/v2.2.0/tensorflow-2.2.0-cp37-cp37m-li  nux_armv7l.whl
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting tensorflow==2.2.0 from https://github.com/leigh-johnson/Tensorflow-bi  n/releases/download/v2.2.0/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl
>   Downloading https://github.com/leigh-johnson/Tensorflow-bin/releases/download/  v2.2.0/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl (109.3MB)
>     100% |████████████████████████████████| 109.3MB 4.0kB/s
> Collecting astunparse==1.6.3 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/2b/03/13dde6512ad7b4557eb7  92fbcf0c653af6076b81e5941d36ec61f7ce6028/astunparse-1.6.3-py2.py3-none-any.whl
> Collecting six>=1.12.0 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/ee/ff/48bde5c0f013094d729f  e4b0316ba2a24774b3ff1c52d924a8a4cb04078a/six-1.15.0-py2.py3-none-any.whl
> Collecting absl-py>=0.7.0 (from tensorflow==2.2.0)
>   Downloading https://www.piwheels.org/simple/absl-py/absl_py-0.9.0-py3-none-any  .whl (121kB)
>     100% |████████████████████████████████| 122kB 455kB/s
> Collecting termcolor>=1.1.0 (from tensorflow==2.2.0)
>   Downloading https://www.piwheels.org/simple/termcolor/termcolor-1.1.0-py3-none  -any.whl
> Collecting grpcio>=1.8.6 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/5e/29/1bd649737e427a6bb850  174293b4f2b72ab80dd49462142db9b81e1e5c7b/grpcio-1.30.0.tar.gz (19.7MB)
>     100% |████████████████████████████████| 19.7MB 23kB/s
> Collecting wheel>=0.26; python_version >= "3" (from tensorflow==2.2.0)
>   Using cached https://files.pythonhosted.org/packages/8c/23/848298cccf8e40f5bbb  59009b32848a4c38f4e7f3364297ab3c3e2e2cd14/wheel-0.34.2-py2.py3-none-any.whl
> Collecting google-pasta>=0.1.8 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/a3/de/c648ef6835192e6e2cc0  3f40b19eeda4382c49b5bafb43d88b931c4c74ac/google_pasta-0.2.0-py3-none-any.whl (57  kB)
>     100% |████████████████████████████████| 61kB 1.9MB/s
> Collecting keras-preprocessing>=1.1.0 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/79/4c/7c3275a01e12ef9368a8  92926ab932b33bb13d55794881e3573482b378a7/Keras_Preprocessing-1.1.2-py2.py3-none-  any.whl (42kB)
>     100% |████████████████████████████████| 51kB 4.2MB/s
> Collecting protobuf>=3.8.0 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/47/90/27eac528bb6871336e86  ed8bcbb72a9b412a4020bf69bb7be71d40e5eab0/protobuf-3.12.2-py2.py3-none-any.whl (4  43kB)
>     100% |████████████████████████████████| 450kB 985kB/s
> Collecting tensorflow-estimator<2.3.0,>=2.2.0 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/a4/f5/926ae53d6a226ec0fda5  208e0e581cffed895ccc89e36ba76a8e60895b78/tensorflow_estimator-2.2.0-py2.py3-none  -any.whl (454kB)
>     100% |████████████████████████████████| 460kB 977kB/s
> Collecting wrapt>=1.11.1 (from tensorflow==2.2.0)
>   Downloading https://www.piwheels.org/simple/wrapt/wrapt-1.12.1-cp37-cp37m-linu  x_armv7l.whl (68kB)
>     100% |████████████████████████████████| 71kB 318kB/s
> Collecting scipy==1.4.1; python_version >= "3" (from tensorflow==2.2.0)
>   Downloading https://www.piwheels.org/simple/scipy/scipy-1.4.1-cp37-cp37m-linux  _armv7l.whl (50.7MB)
>     100% |████████████████████████████████| 50.7MB 9.2kB/s
> Collecting opt-einsum>=2.3.2 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/63/a5/e6c07b08b934831ccb8c  98ee335e66b7761c5754ee3cabfe4c11d0b1af28/opt_einsum-3.2.1-py3-none-any.whl (63kB  )
>     100% |████████████████████████████████| 71kB 4.0MB/s
> Collecting h5py<2.11.0,>=2.10.0 (from tensorflow==2.2.0)
>   Downloading https://www.piwheels.org/simple/h5py/h5py-2.10.0-cp37-cp37m-linux_  armv7l.whl (4.7MB)
>     100% |████████████████████████████████| 4.7MB 98kB/s
> Collecting numpy<2.0,>=1.16.0 (from tensorflow==2.2.0)
>   Downloading https://www.piwheels.org/simple/numpy/numpy-1.19.0-cp37-cp37m-linu  x_armv7l.whl (10.5MB)
>     100% |████████████████████████████████| 10.5MB 33kB/s
> Collecting gast==0.3.3 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/d6/84/759f5dd23fec8ba71952  d97bcc7e2c9d7d63bdc582421f3cd4be845f0c98/gast-0.3.3-py2.py3-none-any.whl
> Collecting tensorboard<2.3.0,>=2.2.0 (from tensorflow==2.2.0)
>   Downloading https://files.pythonhosted.org/packages/1d/74/0a6fcb206dcc72a6da9a  62dd81784bfdbff5fedb099982861dc2219014fb/tensorboard-2.2.2-py3-none-any.whl (3.0  MB)
[... cut ...]
> OSError: libedgetpu.so.1: cannot open shared object file: No such file or directory
> Exception ignored in: <function Delegate.__del__ at 0xa5eb8fa8>
> Traceback (most recent call last):
>   File "/home/pi/.venv/lib/python3.7/site-packages/tensorflow/lite/python/interpreter.py", line 125, in __del__
>     if self._library is not None:
> AttributeError: 'Delegate' object has no attribute '_library'
> (.venv) pi@raspberrypi:~ $ 

Thanks in advance.  I really appreciate your help with this issue...

GBW48, 2020-06-29:
Hi Leigh, I just wanted to let you know that I tried your fix for the Coral runtime error. Unfortunately, the suggested changes did not work. I posted the results on GitHub.

When you have an opportunity, I’d appreciate it if you could look into the Coral runtime issue. Thanks in advance! ...Gary White

------ Original Message ------

From: Leigh Johnson
To: leigh-johnson/rpi-deep-pantilt
Cc: GBW48, Mention
Sent: June 24, 2020 at 10:53 PM
Subject: Re: [leigh-johnson/rpi-deep-pantilt] Error encountered when trying to use Google Coral accelerator (#37)

Thanks for the bug report @GBW48 (https://github.com/GBW48), sorry this didn't work for you!

Looks like your TensorFlow installation didn't finish because you don't have the wheel Python package installed. I think my Coral installation instructions are out of date too. Try the following to re-install, let me know how it goes. 🤞

Let me know how it goes a (with your virtual environment active) to reinstall everything

Install wheel and reinstall TensorFlow

(.venv)$ pip install wheel && pip install --force-reinstall https://github.com/leigh-johnson/Tensorflow-bin/releases/download/v2.2.0/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl

Install/reinstall Coral's libedgetpu.

$ echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list $ curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add - $ sudo apt-get update $ sudo apt-get install libedgetpu1-std

Re-install Coral's tflite_runtime

(.venv)$ pip install https://dl.google.com/coral/python/tflite_runtime-2.1.0.post1-cp37-cp37m-linux_armv7l.whl

GBW48, 2020-07-04:
Hi Leigh,  I reran your instructions for the install/reinstall of Coral's libedgetpu.  This time there no errors.  After a reboot, I was able to successfully  use the Coral accelerator to display and track both objects and faces.  Every thing worked!  

Thanks so much for creating such an excellent, well conceived and documented project!  What a wonderful way to learn about the capabilities of TensorFlow!
...Gary White

P.S.
I'm not sure why the second attempt to apply your fix worked, while the first attempt failed.   I'm including the console log for the second install.  Perhaps, you might be able to figure out the reason for the failure, if you look at the  June 25, 2020 console log associated with the fist installation of the fix. 

 **(.venv) pi@raspberrypi:~ $ echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee** **/etc/apt/sources.list.d/coral-edgetpu.list**
> deb https://packages.cloud.google.com/apt coral-edgetpu-stable main

**(.venv) pi@raspberrypi:~ $ curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -**
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100   653  100   653    0     0   4766      0 --:--:-- --:--:-- --:--:--  4766
> OK

 **(.venv) pi@raspberrypi:~ $ sudo apt-get update**
> Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
> Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
> Get:3 https://packages.cloud.google.com/apt coral-edgetpu-stable InRelease [6,332 B]
> Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
> Get:5 http://archive.raspberrypi.org/debian buster/main armhf Packages [330 kB]
> Get:6 https://packages.cloud.google.com/apt coral-edgetpu-stable/main armhf Packages [1,172 B]
> Fetched 13.4 MB in 7s (2,019 kB/s)
> Reading package lists... Done

**(.venv) pi@raspberrypi:~ $ sudo apt-get install libedgetpu1-std**
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> The following package was automatically installed and is no longer required:
>   rpi-eeprom-images
> Use 'sudo apt autoremove' to remove it.
> The following NEW packages will be installed:
>   libedgetpu1-std
> 0 upgraded, 1 newly installed, 0 to remove and 8 not upgraded.
> Need to get 274 kB of archives.
> After this operation, 624 kB of additional disk space will be used.
> Get:1 https://packages.cloud.google.com/apt coral-edgetpu-stable/main armhf libedgetpu1-std armhf 14.0 [274 kB]
> Fetched 274 kB in 0s (642 kB/s)
> Selecting previously unselected package libedgetpu1-std:armhf.
> (Reading database ... 96755 files and directories currently installed.)
> Preparing to unpack .../libedgetpu1-std_14.0_armhf.deb ...
> Unpacking libedgetpu1-std:armhf (14.0) ...
> Setting up libedgetpu1-std:armhf (14.0) ...
> Processing triggers for libc-bin (2.28-10+rpi1) ...

I then  re-installed Coral's tflite_runtime which also worked.  Here are the results:
**(.venv) pi@raspberrypi:~ $ pip install https://dl.google.com/coral/python/tflite_runtime-2.1.0.post1-cp37-cp37m-linux_armv7l.whl**
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Requirement already satisfied: tflite-runtime==2.1.0.post1 from https://dl.google.com/coral/python/tflite_runtime-2.1.0.post1-cp37-cp37m-linux_armv7l.whl in ./.venv/lib/python3.7/site-packages (2.1.0.post1)
> Requirement already satisfied: numpy>=1.12.1 in ./.venv/lib/python3.7/site-packages (from tflite-runtime==2.1.0.post1) (1.19.0)

GBW48, 2020-07-04:
Sorry, I closed this issue by mistake.

leigh-johnson, 2020-07-25:
@GBW48 Woooo, glad that worked for you Gary! Thanks for including logs of your second go. At a glance, I'm not sure what's different. 🤔 

I'll prioritize getting installation a bit better scripted/automated. Right now, there are a lot of manual steps that make it tedious to reproduce a working environment. I've been getting by with some Ansible playbooks for my own release testing, but they're too crufty and personal to release for general use.

closed 2020-07-25 by leigh-johnson