bitsy-ai/rpi-object-tracking #39: Use a USB Webcam?

arminf82, 2020-07-17:
* rpi-deep-pantilt version: 1.0.0 (2019-12-01)
* Python version: 3.7
* TensorFlow version: 2.2
* Operating System: Raspberry OS

### Description
Is it possible to use a regular USB Webcam? What changes do I have to make to make it run?

Additional question: When I run a coral model on object detection I get a max 8 FPS. Would that be faster using a Pi Cam?

Regards,
Armin

leigh-johnson, 2020-07-25:
Hey @arminf82! I haven't tried a USB web cam. The performance is definitely going to depend on the camera's frame rate, USB 2 vs 3, and how efficiently the camera's driver can serialize data. 

The PiCamera uses a MIPI serial interface, which is specifically designed for high-speed image data transfer. My gut feeling is that it'll have a higher bitrate on the Pi than most USB cameras. If I were trying to compare the two, I'd write a tiny program that benchmarks how fast each method can read/write to a ring buffer. 
https://www.mipi.org/specifications/csi-2

You could also use cprofile to see if the inference loop is waiting on buffer being available to read. 

Here's the line to replace if you implement a USB image stream. Replace this .read() method with one that takes the most recent frame from your buffer, then zeroes the buffer.
https://github.com/leigh-johnson/rpi-deep-pantilt/blob/master/rpi_deep_pantilt/detect/camera.py#L93

leigh-johnson, 2020-07-25:
Closing for now, so it's easier for me to see issues in need of triage at a glance. Good luck with this!

closed 2020-07-25 by leigh-johnson

arminf82, 2020-08-13:
I bought a PiCam v2. I had two different USB Cams and both of them were much slower.  Thank you.