bitsy-ai/rpi-object-tracking #48: PID controller improvements

leigh-johnson, 2020-10-19:
### Summary

Currently, the PID controller implements basics from Adrian Rosebrock's blog post [Pan/tilt face tracking with a Raspberry Pi and OpenCV](https://www.pyimagesearch.com/2019/04/01/pan-tilt-face-tracking-with-a-raspberry-pi-and-opencv/). The PID gains are hard-coded, roughly tuned to track a desk-height object no further than a few meters. 

### Work Required

* Implement Config API + serializer
* Replace hard-coded PID controller gains with values read from config (default: current gains)
* Add calibration routines for configurable variables (rotation, Px+Ix, Dx, Py+Iy, Dy)
* Implement a ResetBehavior API, with a few common examples like reset x/y to origin, seek neighborhood, seek grid

Martin2kid, 2020-10-19:
Great!!!

That small noisy RC-servo is notorious for fine positional control with quite bit of lagging.
Have you thought about using direct driven brushless motor FOC control like here?https://youtu.be/TnECJG2mN-E
or this at https://simplefoc.com

I like Pablo's 2 motor controller approach over SimpleFOC's single and ordered couple for experiment.

Also, Justine at https://youtu.be/OZvjfbpXpro has dual controller but not sure this support magnetic encoder (cheap & quite accurate) and what I wanted to use. 

All of them shared their code & I hope I can figure ways to drive PWM or I2c (Newbie here)

br0kej, 2020-10-20:
Hello,

I came across this repo a couple of days ago and checked the issues! I am not sure how useful this will be but I saw a post on Reddit today of a Rpi/Arduino powered nerf gun aimbot which seems to be using similar ideas and seems very smooth. When I downloaded and setup some of the examples, the face tracking was pretty rough, especially at close range. Hopefully this could be of use to support the development of the features above - [Github Link](https://github.com/DDeGonge/Nerf_Aimbot) / [Reddit Link](https://www.reddit.com/r/Nerf/comments/je2o9f/i_made_a_nerf_aimbot_fully_3d_printed_and_pi/)

Martin2kid, 2020-10-20:
Hi,

Have you tried with brushless motor with higher poll count over 12 & not stepper? 
I tried with 32 poll motor with Simplebgc32 controller that uses same control method here and it provided very smooth rotation in silence; 
https://youtu.be/Ce-c9StqzsE

The problem that I had with such IMU based controller is that it drift(slowly turn) due to gyro drift even with magnetometer and also main reason I was looking for one without IMU

Martin2kid, 2020-11-04:
Leigh,

Per closed issue of #50, I've been thinking about how your setup & mine were producing different result, although we both have same setup of Pimorini pan tilt hat & same camera orientation.(assuming your setup were working with resume tracking of face when it return to FOV) but mine didn't,  even though I'm using Buster 8-20-2020.

I thought about typical servo motor that Pimorini used on this hat that each batch of production run for this kind of brushed motor servo may have different gear mesh tolerance, motor output torque, motor's brush resistance, gear friction tolerance level & lubrication density that in-turn, may have affected PID rate, thus I suspect may have produced totally different outcome.

Alternatively, use of general but better quality brushless motor servo may provide slightely better PID control output but will more-likely end up with costing annoying high frequency pitched noise & same gear meshing sound.

For programmers perspective, I'd assume this will create rather unpredictable outcome of static and general working PID values that in turn require self calibration of Pan&tilt hat mechanical PID estimation, storing that calibration data and fused that data with Tf's output of detection flow.

I don't think, typical PWM driven servo can produce such PID feed back info to controller like I2c type and look forward to see parallel code implementation including I2c (I think it's also possible for 360 degree & beyond continuous tracking, faster angle position acqusiton, self-error correction and ROS implementation feasibility.

Main reason newbie is continuously looking into FOC motor control method and keep studying.

Martin2kid, 2020-11-06:
Leigh,

I've been investigated my Pi 4's particular behavior which is now closed issue of #50 that RPI hangs when face goes out of FOV but maintain face recognition but fail to resume face tracking.

For possible solution & investigation to find out cause of nature, I also tried with OpenVINO using NCS2 with same Adrian's PID setup per this https://github.com/squeakus/bitsandbytes/blob/master/facetrack/deep_tracking.py
and came up with same behavior result that Pi 4 hangs(camera goes extreme up, not returning to mid-position, still does face recognition but fails to resume tracking).---shockingly similar result.

I'm wondering your setup was tested for this scenario (subject face goes out of FOV & return and Pi4 resume tracking)?
Your response implying yours may be working correct despite my  issue reporting in both #50 & #48, and advised me to check camera orientation & PID setting which as I indicated is same as your setup or per given instruction over Web.

I'm beginning to think Adrian's PID implementation method, and adaptive usage with Tf or OpenVINO may have something to do with this issue.

I know jp3spinelli  has researched this or similar cause of issue extensively but did not hear from him in issue #44 that I've commented.

OpenVINO code that resulted same hangs are as follows for your reference(main code only here but see github link above for full info);

# import necessary packages
from multiprocessing import Manager, Process
from imutils.video import VideoStream
from utils.deepcenter import ObjCenter
from utils.pid import PID
#from pyimagesearch.objcenter import ObjCenter
#from pyimagesearch.pid import PID
import pantilthat as pth
import argparse
import signal
import time
import sys
import cv2

#RESOLUTION = (320, 320)

# define the range for the motors
servoRange = (-90, 90)

# function to handle keyboard interrupt
def signal_handler(sig, frame):
    # print a status message
    print("[INFO] You pressed `ctrl + c`! Exiting...")

    # disable the servos
    pth.servo_enable(1, False)
    pth.servo_enable(2, False)

    # exit
    sys.exit()

def obj_center(args, objX, objY, centerX, centerY):
    # signal trap to handle keyboard interrupt
    signal.signal(signal.SIGINT, signal_handler)

    # start the video stream and wait for the camera to warm up
    vs = VideoStream(usePiCamera=True).start()
    time.sleep(2.0)

    # initialize the object center finder
    #obj = ObjCenter(args["cascade"])#original from Adrian
    obj = ObjCenter(args["model"])

    # loop indefinitely
    while True:
        # grab the frame from the threaded video stream and flip it
        # vertically (since our camera was upside down)
        frame = vs.read()
        frame = cv2.flip(frame, 0)

        # calculate the center of the frame as this is where we will
        # try to keep the object
        (H, W) = frame.shape[:2]
        centerX.value = W // 2
[... 99 lines cut ...]
        # 4. setServos     - drives the servos to proper angles based
        #                    on PID feedback to keep object in center
        processObjectCenter = Process(target=obj_center,
            args=(args, objX, objY, centerX, centerY))
        processPanning = Process(target=pid_process,
            args=(pan, panP, panI, panD, objX, centerX))
        processTilting = Process(target=pid_process,
            args=(tlt, tiltP, tiltI, tiltD, objY, centerY))
        processSetServos = Process(target=set_servos, args=(pan, tlt))

        # start all 4 processes
        processObjectCenter.start()
        processPanning.start()
        processTilting.start()
        processSetServos.start()

        # join all 4 processes
        processObjectCenter.join()
        processPanning.join()
        processTilting.join()
        processSetServos.join()

        # disable the servos
        pth.servo_enable(1, False)
        pth.servo_enable(2, False)

Martin2kid, 2020-12-04:
Leigh,

This is just suggestion to incorporate Pan & Tilt device capable of rotating beyond 180 degree angle limitation.
  
Can there be anyway to reset PWM signal within processing so that center of FOV image can be used or refreshed by processing thus able to handle Pan & Tilt device capable of 360 degree rotation & beyond such as 360 degree winch servo and other brushless gimbal accepting PWM signal input?

Martin2kid, 2021-02-12:
Leigh,

I've tried to make FOC BLDC controller(not IMU based Gimbal controller such as; Simplebgc32 or Storm32 due to gyro drift) to work with rpi-deep-pantilt & none of developing platform based on Arduino or STM32 supporting PWM input except Odrive that is rather pricy ($120 plus)
 
After several month out of head scratching, I finally ordered couple of Drive which I'm sure is overkill, yet it support PWM input & also provide 360 degree magnetic absolute sensor support such as AS5048a.

I'll update outcome here after test drive with Software PID & Combined with both hardware PID control loop.

Hoping you would find solution for out of FOV behavior soon!