bitsy-ai/rpi-object-tracking #52: Multiple Issues ...

Moehrenbaum, 2020-11-29:
I was so excited when I found this demo and still you did a wonderful job! It's just that I struggle to get the parts to work.

It seems since you started this project there has been many updates in many areas and nothing really matches anymore e.g. description in the blog and various versions of code and frameworks. Sorry, just my best guess. I'm new to raspi and ML on raspi with CORAL -- latter tech is cool but it appears more demos don't work for a newbie. 

Hours in the only thing that works according to script is : **rpi-deep-pantilt test pantilt**. The thing moves! Yes.

As for the next one : **rpi-deep-pantilt test camera** it's not crashing but I also can't see anyting. Despite the fact that I have turned on raspi cam and rebooted .... I see the desktop on my TV.

**rpi-deep-pantilt detect --edge-tpu** seems to load the labels (shows is it but ...) but presents this afterwards. I guess some error

> 2020-11-29 16:36:18.764478: W tensorflow/core/framework/cpu_allocator_impl.cc:80] Allocation of 307200 exceeds 10% of free system memory.
2020-11-29 16:36:22.000636: W tensorflow/core/framework/cpu_allocator_impl.cc:80] Allocation of 307200 exceeds 10% of free system memory.
2020-11-29 16:36:22.023519: W tensorflow/core/framework/cpu_allocator_impl.cc:80] Allocation of 307200 exceeds 10% of free system memory.
2020-11-29 16:36:22.072725: W tensorflow/core/framework/cpu_allocator_impl.cc:80] Allocation of 307200 exceeds 10% of free system memory.
2020-11-29 16:36:22.093627: W tensorflow/core/framework/cpu_allocator_impl.cc:80] Allocation of 307200 exceeds 10% of free system memory.

Now, this is only after I installed the **latest ftlite framework** from the Coral website. Prior to this there were other issues.

I understand you are doing this in your spare time and I truly appreciate you sharing this. Please don't take the text above as a rant. By no means is it meant that way. 

I'm just wondering if there is some help you can offer on how to check if the correct libs are in place .... and how to make this work. I'm almost 3 days in with fiddling and now want to see it work.

Thanks in advance, Dirk

p.s. your blog still says:
**pip install https://github.com/leigh-johnson/Tensorflow-bin/blob/master/tensorflow-2.0.0-cp37-cp37m-linux_armv7l.whl\?raw\=true**

which leads into this error:

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting tensorflow==2.0.0
  ERROR: HTTP error 404 while getting https://github.com/leigh-johnson/Tensorflow-bin/blob/master/tensorflow-2.0.0-cp37-cp37m-linux_armv7l.whl?raw=true
ERROR: Could not install requirement tensorflow==2.0.0 from https://github.com/leigh-johnson/Tensorflow-bin/blob/master/tensorflow-2.0.0-cp37-cp37m-linux_armv7l.whl?raw=true because of HTTP error 404 Client Error: Not Found for url: https://github.com/bitsy-ai/tensorflow-arm-bin/blob/main/tensorflow-2.0.0-cp37-cp37m-linux_armv7l.whl for URL https://github.com/leigh-johnson/Tensorflow-bin/blob/master/tensorflow-2.0.0-cp37-cp37m-linux_armv7l.whl?raw=true

Martin2kid, 2020-12-02:
Moehrenbaum,

You are correct that 10% CPU memory is just warning & it has nothing to do with rpi-deep-pantilt and you can ignore that.

I'm newbie too, obviously you saw Legh's post at Toward Data Science as you stated "pip install https://github.com/leigh-johnson/Tensorflow-bin/blob/master/tensorflow-2.0.0-cp37-cp37m-linux_armv7l.whl\?raw\=true".

You are leaving this message at Github that actually specify "$ pip install https://github.com/leigh-johnson/Tensorflow-bin/releases/download/v2.2.0/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl" which I don't think it is same version as you installed(Leigh stated it is "community-built wheel" which imply it is special version).

I also noticed it takes multiple attempts to Pip install that file but I was able to install in 4th try about few weeks ago.

If you are using Coral USB, follow in step 12 in Toward Data Science blog.

Cheers!

Moehrenbaum, 2020-12-05:
> Moehrenbaum,
> 
> You are correct that 10% CPU memory is just warning & it has nothing to do with rpi-deep-pantilt and you can ignore that.
> 
> I'm newbie too, obviously you saw Legh's post at Toward Data Science as you stated "pip install [https://github.com/leigh-johnson/Tensorflow-bin/blob/master/tensorflow-2.0.0-cp37-cp37m-linux_armv7l.whl\?raw\=true](https://github.com/leigh-johnson/Tensorflow-bin/blob/master/tensorflow-2.0.0-cp37-cp37m-linux_armv7l.whl%5C?raw%5C=true)".
> 
> You are leaving this message at Github that actually specify "$ pip install https://github.com/leigh-johnson/Tensorflow-bin/releases/download/v2.2.0/tensorflow-2.2.0-cp37-cp37m-linux_armv7l.whl" which I don't think it is same version as you installed(Leigh stated it is "community-built wheel" which imply it is special version).
> 
> I also noticed it takes multiple attempts to Pip install that file but I was able to install in 4th try about few weeks ago.
> 
> If you are using Coral USB, please follow in step 12 in Toward Data Science blog.
> 
> I've also noticed that discrepancy between Toward & Github post but I must still give her all the credit for sharing this code with us and it is up to us carefully try one post after another or combination and further reading through all available issue posted including closed post.
> 
> Cheers!

Martin2kid, thanks again for helping out. I got it all working. I guess I simplified to my skill level :) AND a big thanks to Leigh for sharing her awesome work.