bitsy-ai/rpi-object-tracking #53: Where to change line_thickness. Seems to be defaulted to 4?

Moehrenbaum, 2020-12-03:
I'm puzzled. I got the code working. All good. However, for some reason I cannot change the line_thickness. None of my change seem to make any difference. It appears to be defaulted around 4.

What am I missing? Sorry, maybe a noobs question but we all start from a place of comfort.

Thx

Martin2kid, 2020-12-04:
Moehrenbaum,

Great you have it working now, I think what your looking for is here
pi_deep_pantilt/detect/util/visualization.py

hope that help

Moehrenbaum, 2020-12-05:
> Moehrenbaum,
> 
> Great! you have it working now, I think what your looking for is here;
> pi_deep_pantilt/detect/util/visualization.py
> 
> I would recommend to make sure clone the SD card & safe keeping it as it is difficult to replicate same install (Tflite changes & other variable changes)
> 
> hope that help

Thanks Martin2kid, for responding. Turned out my code was always using parts from the site-package and not from inside the package. I must have done a pip install rpi-deep-pantilt into my virtual env .... 

After I uninstalled latter I got it to work, after changing import statements and a path setting. 

I guess all of this is clear to someone with better Python skills but that's not me. 

So, again thanks for replying!

Martin2kid, 2020-12-05:
Moehrenbaum,

Glad I could be a help!

If you want it start automatically (every time you turn Pi on)

1st modify this file:  /etc/xdg/lxsession/LXDE-pi/autostart #you'll need to use sudo nano to edit this file
@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@lxterminal -e "/home/pi/auto-start.sh" #this new line, auto-start.sh file need to be created first in this folder & chmod that file
@xscreensaver -no-splash

My auto-start.sh file content

#!/bin/bash
cd rpi-deep-pantilt
source .venv/bin/activate
rpi-deep-pantilt track --edge-tpu face