bitsy-ai/rpi-object-tracking #55: Failed to write byte Error

jiyoonpark0207, 2021-02-06:
I followed all the instructions but keep on getting this error. Can someone help me? 
```
(.venv) pi@raspberrypi:~ $ rpi-deep-pantilt test pantilt

-- output 
![Screenshot from 2021-02-06 22-25-46](https://user-images.githubusercontent.com/49419698/107119346-4da0d300-68ca-11eb-9457-f15262814032.png)

```
it says OSError:  Failed to write byte
does anyone know how to fix this?

Martin2kid, 2021-02-12:
jiyoon,

Are you using Pimorini pan-tilt hat?

Sreehar, 2022-05-12:
> I followed all the instructions but keep on getting this error. Can someone help me?
> 
> ```
> (.venv) pi@raspberrypi:~ $ rpi-deep-pantilt test pantilt
> 
> -- output 
> 
> (.venv) pi@raspberrypi:~ $ rpi-deep-pantilt test pantilt
> INFO:root:Starting Pan-Tilt HAT test!
> INFO:root:Pan-Tilt HAT should follow a smooth sine wave
> Traceback (most recent call last):
>   File "/home/pi/.venv/bin/rpi-deep-pantilt", line 8, in <module>
>     sys.exit(main())
>   File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/cli.py", line 153, in main
>     cli()
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 829, in __call__
>     return self.main(*args, **kwargs)
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 782, in main
>     rv = self.invoke(ctx)
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/home/pi/.venv/lib/python3.7/site-packages/click/core.py", line 610, in invoke
>     return callback(*args, **kwargs)
>   File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/cli.py", line 141, in pantilt
>     return pantilt_test()
>   File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/control/hardware_test.py", line 38,
>     pantilthat.pan(a)
>   File "/home/pi/.venv/lib/python3.7/site-packages/pantilthat/pantilt.py", line 466, in servo_one
>     self.setup()
>   File "/home/pi/.venv/lib/python3.7/site-packages/pantilthat/pantilt.py", line 80, in setup
>     self._set_config()
>   File "/home/pi/.venv/lib/python3.7/site-packages/pantilthat/pantilt.py", line 118, in _set_config
>     self._i2c_write_byte(self.REG_CONFIG, config)
>   File "/home/pi/.venv/lib/python3.7/site-packages/pantilthat/pantilt.py", line 209, in _i2c_write_byte
>     raise IOError("Failed to write byte")
> OSError: Failed to write byte
> ```
> 
> it says OSError: Failed to write byte does anyone know how to fix this?

Did you fix the issue ?
If you did, how was it done?
I get the same thing as error .
Traceback (most recent call last):
  File "/home/pi/track/facetracker.py", line 37, in <module>
    pan(cam_pan-90)
  File "/usr/lib/python3/dist-packages/pantilthat/pantilt.py", line 466, in servo_one
    self.setup()
  File "/usr/lib/python3/dist-packages/pantilthat/pantilt.py", line 80, in setup
    self._set_config()
  File "/usr/lib/python3/dist-packages/pantilthat/pantilt.py", line 118, in _set_config
    self._i2c_write_byte(self.REG_CONFIG, config)
  File "/usr/lib/python3/dist-packages/pantilthat/pantilt.py", line 209, in _i2c_write_byte
    raise IOError("Failed to write byte")
OSError: Failed to write byte