bitsy-ai/rpi-object-tracking #57: rpi-deep-pantilt track face

adelorenzo, 2021-05-22:
* rpi-deep-pantilt version: 1.21
* Python version: 3.7.3
* TensorFlow version: 2.4.0
* Operating System: raspbian buster

### Description

When I run  `rpi-deep-pantilt track face --edge-tpu` or `rpi-deep-pantilt track face` I get the following error:

```
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/camera.py", line 26, in run_pantilt_detect
    model = model_cls()
  File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/facessd_mobilenet_v2.py", line 74, in __init__
    self.PATH_TO_LABELS, use_display_name=True)
  File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/util/label.py", line 172, in create_category_index_from_labelmap
    label_map_path, use_display_name)
  File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/util/label.py", line 132, in create_categories_from_labelmap
    label_map = load_labelmap(label_map_path)
  File "/home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/detect/util/label.py", line 105, in load_labelmap
    label_map_string = fid.read()
  File "/home/pi/.venv/lib/python3.7/site-packages/tensorflow/python/lib/io/file_io.py", line 117, in read
    self._preread_check()
  File "/home/pi/.venv/lib/python3.7/site-packages/tensorflow/python/lib/io/file_io.py", line 80, in _preread_check
    compat.path_to_str(self.__name), 1024 * 512)
tensorflow.python.framework.errors_impl.NotFoundError: /home/pi/.venv/lib/python3.7/site-packages/rpi_deep_pantilt/data/facessd_label_map.pbtxt; No such file or directory
```
Every other label works perfectly. Kindly advise.

Thank you.

adelorenzo, 2021-05-22:
I upgraded to version 2.0.0 and it started working. Thank you!!

closed 2021-05-22 by adelorenzo

leigh-johnson, 2021-05-22:
Thank you for the report! Let me know if you ran into documentation that prompted you to install 1.x - I'll make sure this gets updated.

adelorenzo, 2021-05-22:
Hello Leigh,

I followed the installation steps on github basically. After some digging I found v2.0.0 and decided to give it a shot, and it worked! It would be cool if the output of the framebuffer could be streamed via HTTP. If you have any idea of how this could be achieved please let me know.

I work with video/blog production and your project would be perfect for this purpose. Thank you so much.

Best regards,
Adolfo Delorenzo

careyer, 2021-09-26:
This very same problem occured to me as well... can you please describe how to slove that problem ... i cannot comprehend what you mean with "update to 2.0.0".  Thanks!