bitsy-ai/rpi-object-tracking #58: stream the output to mjpegstreamer or similar?

adelorenzo, 2021-05-22:
Sorry to put this question here but is there a way to stream the output for instance with mjpegstreamer or similar?

Thank you.

leigh-johnson, 2021-05-22:
Thanks for this idea, love it! Can you say a little more about how you would be using the video stream?

A few notes for myself:

The picamera library provides a streaming example here.
 https://picamera.readthedocs.io/en/release-1.13/recipes2.html?highlight=mjpeg#web-streaming

Compositing an mmal overlay might "just work" with this fork of mjpeg-streamer, which supports RaspiCam's mmal as an input source. This fork is used in OctoPrint (web interface for 3D printers)
https://github.com/jacksonliam/mjpg-streamer
https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_raspicam/input_raspicam.c

adelorenzo, 2021-05-22:
I work with video/blog production and my idea is to stream live via OBS using the output of rpi-object-tracking. I'll take a look at docs and let you know how it goes.
Thank you!!